elhep/SpaceVPX_FMC_Carrier_3U #5: LPDDR

mkuklewski, 2019-02-07:
There are two issues here:
1.  Resistor connected to D7 MDDR_IMP_CALIB (FPGA_Config_SchDoc) should be 150 Ohm (according to Pin_Descripions_Datasheet),
2. Two pins on MT46H64M32LFBQ-48 should be replaced - F2 with E2.

tprzywoz, 2019-03-05:
done

tprzywoz, 2019-04-02:
I'm confused. According to [DS0115](https://www.microsemi.com/document-portal/doc_view/130718-ds0115-smartfusion2-pin-descriptions-datasheet) D7 MDDR_IMP_CALIB resistor should be 150 Ohm and clk lines don't have termination resistor. According to [AC393](https://www.microsemi.com/document-portal/doc_view/129979-ac393-board-and-layout-design-guidelines-for-smartfusion2-soc-and-igloo2-fpgas-application-note) it should be 240 Ohm and there is a 100 Ohm termination at the end of clk lines. M2S090TS-EVAL-KIT is designed according to AC393, although it has same memory chip as DS0115 example.

mkuklewski, 2019-04-02:
In my opinion there is something not mentioned inside AC393 and its configuration (for example IO calibration is off) or there is an error in documentation. I would use 150 Ohms resistor (same voltage as DDR2 interface, which is also not done correctly in this document).
As for 100 Ohm resistor, it should be included, because MT46H64M32LFBQ CLK inputs are differential.

tprzywoz, 2019-04-14:
I used 150 Ohm resistor + added 100 Ohm resistor at LPDDR's CLK inputs

closed 2019-04-14 by tprzywoz