elhep/SpaceVPX_FMC_Carrier_3U #10: PCB review

tprzywoz, 2019-06-17:


mkuklewski, 2019-06-26:
Ok, thank you for your time :)

Here are couple issues - which justification can be found in the ECSS-Q-ST-70-12C(14July2014).pdf:
0. All vias should be exposed, if it is possible - because we are using solder mask (although it shouldn't be used) and all traces are between are hidden between top and bottom layers, it is very hard to access them (like under BGA) and debug if something is not right.
1. If it is possible please change via's annual ring size to 100um,
2. **Please expose top and bottom parts of the PCB which are used as thermal interface for wedge locks (NIC design),**
3. No copper balancing on TOP and BOTTOM layers,
4. **Power planes on layer from L2 to L4 - they should be spilled over whole PCB to ensure homogeneous copper distribution and proper heat distribution (please check NIC design) - but not connected to TOP and BOTTOM layers**
5. Grid should be offset between layers and should be placed symmetrically within the build-up to minimise warp and twist.
6. Tear drop reinforcement of pads to tracks can be used for a more robust design and to mitigate the risk of a smaller designed diameter.

Here are some other issues:
1. Please include test points for: all power rails, reset signal, SpaceWire_A, PROBE_A and _B (W10 and Y10 pins) (I would like to add more, but there is already no space )
2. There are no fiducial markers
3. **Power Good lines (and PWR_OVLn) are connected directly to MCU  - please connected them trough resistor divider - those lines are pulled up to P12V0**
4. Connect CLK_DIR to configurable resistor - at least for now (should be configured that carrier is master)
5. I think for GBT we need to choose default configuration - probably it will be better to connect them to UltraThin Pipes Data
6. FRAM PWR A can be deleted

I guess that's all for now

gkasprow, 2019-06-26:
Please make a dedicated issue and copy [this](https://github.com/sinara-hw/sinara/wiki/PCBReviewChecklist?fbclid=IwAR28e8ZezI5MR6vIaSIVShmZltHDKoElIAUoUdAOtR-R-vOO_gkPNsD2J6o) and https://github.com/sinara-hw/sinara/wiki/PCBReviewChecklist?fbclid=IwAR28e8ZezI5MR6vIaSIVShmZltHDKoElIAUoUdAOtR-R-vOO_gkPNsD2J6o checklists. Mark every item once it is verified

gkasprow, 2019-06-26:
I updated the links to the checklists.

gkasprow, 2019-06-26:
a few remarks

> Power planes on layer from L2 to L4 - they should be spilled over whole PCB to ensure homogeneous copper distribution and proper heat distribution (please check NIC design) - but not connected to TOP and BOTTOM layers

this is not a good idea. it's much better to use copper balancing. Solid, not connected polygon acts as an antenna and causes unwanted couplings.

> 
> All vias should be exposed, if it is possible - because we are using solder mask (although it shouldn't be used) and all traces are hidden between top and bottom layers, it is very hard to access them (like under BGA) and debug if something is not right.

Only certain vias should be exposed. Vias must be exposed either on both sides or not at all.
It's better to use 1mm test pads. Vias as test points are poor because can be damaged with a single touch of the probe. Exposed vias under BGA are asking for troubles.
So expose only vias which must be esposed, i.e. under the thermal pads.

we are using solder mask, so why not to use copper balancing on top and bottom?

mkuklewski, 2019-06-27:
> 
> 
> Please make a dedicated issue and copy [this](https://github.com/sinara-hw/sinara/wiki/SchematicsReviewChecklist?fbclid=IwAR1BNygVAagBgiaVf0A0Y6eGzzE2ItLHPzbkBdk3sft3B9_OMpTz3JWIBnM) and [this](https://github.com/sinara-hw/sinara/wiki/PCBReviewChecklist?fbclid=IwAR28e8ZezI5MR6vIaSIVShmZltHDKoElIAUoUdAOtR-R-vOO_gkPNsD2J6o) checklists. Mark every item once it is verified

I have started doing it according to the PCB checklist and I am a halfway through.

> >Power planes on layer from L2 to L4 - they should be spilled over whole PCB to ensure homogeneous copper distribution and proper heat distribution (please check NIC design) - but not connected to TOP and BOTTOM layers

>this is not a good idea. it's much better to use copper balancing. Solid, not connected polygon acts as an antenna and causes unwanted couplings.

I did not meant big solid, not connected polygons, but ground and power planes (in form of grid) which are already there - but increase their size to match PCB size. If they wont be placed then our thermal interface will be very poor.
This is how SpaceVPX_FMC_Carrier looks right now (without TOB and BOTTOM layers):
[image]

[image]

[image]

And this is how NIC design looks:
[image]

[image]

> Only certain vias should be exposed. Vias must be exposed either on both sides or not at all.
> It's better to use 1mm test pads. Vias as test points are poor because can be damaged with a single [...]

Ok, then please increase number of test points. Although we really should start thinking about prototyping PCBs without solder mask...

> we are using solder mask, so why not to use copper balancing on top and bottom?

I have pointed it out as one of the issues - that there is no copper balancing on top and bottom.

mkuklewski, 2019-06-27:
Also I am wondering if this distance between plane and differential line is ok? 
[image]

gkasprow, 2019-06-27:
no, that's not correct. Such low distance will affect trace impedance. Please increase clearance at least to 15mils.

tprzywoz, 2019-07-04:
> 1. Please include test points for all power rails, reset signal, SpaceWire_A, PROBE_A and _B (W10 and Y10 pins) (I would like to add more, but there is already no space )

Do you need test points for both differential lines of SpaceWire? There isn't much space and it would be easier with 4 TP.

gkasprow, 2019-07-04:
For SpW we have already SVPX connectors pads.

tprzywoz, 2019-07-04:
> 4. Connect CLK_DIR to configurable resistor - at least for now (should be configured that carrier is master)

What is the configuration of a master carrier? In standard FMC that pin is floating.

> 5. I think for GBT we need to choose default configuration - probably it will be better to connect them to UltraThin Pipes Data

Is this the final decision? I added UltraThin Pipes Data as an alternative connection - "old default" was FMC connection.

gkasprow, 2019-07-04:
Do you mean CLK DIR in FMC? It is sensed by carrier. We don't really care because our board is LPC so the CLK DIR has no function. 
For GBT we need to setup connection between FPGA and Jetson PCIe. UTP has only single lane.
FMC is one option, UTP is another.

mkuklewski, 2019-07-05:
> Do you mean CLK DIR in FMC? It is sensed by carrier. We don't really care because our board is LPC so the CLK DIR has no function.

Ok, I thought that CLK DIR was set by the carrier to inform the FMC about the direction of the clock. 

> For GBT we need to setup connection between FPGA and Jetson PCIe. UTP has only single lane.
> FMC is one option, UTP is another.

I guess our first usage will be most likely connecting FPGA to Jetson PCIe, rather than to FMC :)
Anyway - if it is configurable, we can always change it.

tprzywoz, 2019-07-09:
> 1. All vias should be exposed, if it is possible - because we are using solder mask (although it shouldn't be used) and all traces are hidden between top and bottom layers, it is very hard to access them (like under BGA) and debug if something is not right.

>Only certain vias should be exposed. Vias must be exposed either on both sides or not at all.

Only vias in thermal pads (power IC) are exposed.

> 2. If it is possible please change via's annual ring size to 100um,

All vias annual ring changed to 100 um.

> 3. **Please expose top and bottom parts of the PCB which are used as thermal interface for wedge locks (NIC design),**

done
> 4. No copper balancing on TOP and BOTTOM layers,

done
> 5. **Power planes on layer from L2 to L4 - they should be spilled over whole PCB to ensure homogeneous copper distribution and proper heat distribution (please check NIC design) - but not connected to TOP and BOTTOM layers**

done
> 6. Grid should be offset between layers and should be placed symmetrically within the build-up to minimise warp and twist.

done
> 7. Tear drop reinforcement of pads to tracks can be used for a more robust design and to mitigate the risk of a smaller designed diameter.

done

> 1. Please include test points for: all power rails, reset signal, SpaceWire_A, PROBE_A and _B (W10 and Y10 pins) (I would like to add more, but there is already no space )

All added except SpaceWire_A

> 2. There are no fiducial markers

Added
> 3. **Power Good lines (and PWR_OVLn) are connected directly to MCU  - please connected them trough resistor divider - those lines are pulled up to P12V0**

Added 3,6 kOhms to GND for 10 k pullup and 24 kOhms to GND for 68 k pullup. I changed also net label - rail P1V8 has P1V5 Power Good net label.
> 4. Connect CLK_DIR to configurable resistor - at least for now (should be configured that carrier is master)

Without changes

> 5. I think for GBT we need to choose default configuration - probably it will be better to connect them to UltraThin Pipes Data

Created default variant with connection to UltraThin Pipes Data

> 6. FRAM PWR A can be deleted

Deleted

tprzywoz, 2019-07-10:
> Also I am wondering if this distance between plane and differential line is ok?

Plane to differential line distance increased to 0.4 mm. 
I left only distance between plane and via's of differential line to prevent power plane from cutting under the FPGA and FMC.

mkuklewski, 2019-07-16:
1.	There is no keep-out lines on the central part of the PCB. I think we should include them. This will limit size of installed FMC’s shield.
[image]
 

2.	The distance from the via must be increased – via itself cannot be moved due to the FMC standard. This means that we will need to reduce the size of the board’s shield that will be installed here (because of DRC and keep-out).  
 
[image]

3.	TVS diodes should be placed much closer to the SpaceVPX connector – not under protected IC (all of them) – impedance issue
4.	Inside PCB Checklist components and test pads should be placed min. 200 mils from the EDGE – but there is no place for this. Probably the best would be to create 10-20mils keep-out lines and check if it is ok. The same goes for the traces (those should be min. 20mils).
5.	Top overlay should be read only in one or two directions,
6.	PCIE signals are too close to each other on Layer 3:
 
[image]

7.	Same problem with signals on L1
 
[image]

8.	PCIE links go over different power planes – I guess there is not enough space to route it differently?

Ok, couple simple issues:
10.	Tear drop reinforcement are placed only at one side of the line- I am not sure how to solve this
 
[image]

11.	Change the polygon boundaries in such a way that they overlap with the inner layers - this is mainly about TOP and BOTTOM, on the left side of the PCB.
   
[image]
[image]

12.	One name of the IC’s is placed on solder pads
         
[image]
[image]
[image]
[image]
[image]
[image]

 
Is it possible to enable DRC which will check if soldermaks is placed directly on copper?

13.	Please add No ERC directives to FPF32701MX – PGOOD pin and LM339AD output pin so there will be no error on compilation.

14.	Please sign Probe A and B signal – it can be just “A” and “B”
[image]

 
15.	Please delete “a b c …”
 
[image]

16.	Could you change those test point to the same type as on the other side of the PCB and sign them? Like “TP1”, “TP2”, etc.
17.	No company logo, copyright notice, date code, pcb number, pcb revision, assembly revision blank on silkscreen, serial number blank on silkscreen
18.	Same fiducial markers are used – not really an issue
19.	No stackup information on Layer 1 or drill leyer
20.	Is there a need to leave those GND vias unmasked on bottom layer?
 
[image]

21.	Hole tolerances are not noted anywhere

tprzywoz, 2019-07-24:
> 1. There is no keep-out lines on the central part of the PCB. I think we should include them. This will limit size of installed FMC’s shield.

added

> 2. The distance from the via must be increased – via itself cannot be moved due to the FMC standard. This means that we will need to reduce the size of the board’s shield that will be installed here (because of DRC and keep-out).

Fixed.  Board Cutout was reduced

> 3. TVS diodes should be placed much closer to the SpaceVPX connector – not under protected IC (all of them) – impedance issue

Replaced  D21, D9, D16, D22A, D23A

> 4. Inside PCB Checklist components and test pads should be placed min. 200 mils from the EDGE – but there is no place for this. Probably the best would be to create 10-20mils keep-out lines and check if it is ok. The same goes for the traces (those should be min. 20mils).

With new rules there is min 5 mm (about 20mils) distance from the edge.

> 5. Top overlay should be read only in one or two directions,

Fixed 

> 6 & 7 . PCIE signals are too close to each other on Layer 3:

fixed

> 8. PCIE links go over different power planes – I guess there is not enough space to route it differently?

It seems to me that it is impossible to do that without an addidtional layer.

> Ok, couple simple issues:
> 10.	Tear drop reinforcement are placed only at one side of the line- I am not sure how to solve this

I found out that some of tear drops disappear when I put keep-out grid to generate copper balancing polygon. Now with different action order, it should be ok.

> 11. Change the polygon boundaries in such a way that they overlap with the inner layers - this is mainly about TOP and BOTTOM, on the left side of the PCB.

done

> 12. One name of the IC’s is placed on solder pads
> Is it possible to enable DRC which will check if soldermaks is placed directly on copper?

Ic15 switched off. I added DRC rule and fixed all collision errors.

> 13. Please add No ERC directives to FPF32701MX – PGOOD pin and LM339AD output pin so there will be no error on compilation.

done

> 14. Please sign Probe A and B signal – it can be just “A” and “B”

done

> 15. Please delete “a b c …”

done

> 16. Could you change those test point to the same type as on the other side of the PCB and sign them? Like “TP1”, “TP2”, etc.

done

> 17. No company logo, copyright notice, date code, pcb number, pcb revision, assembly revision blank on silkscreen, serial number blank on silkscreen

On silkscreen added board name, version, date. On schematics WUT logo, copyright notice, date, pcb name, pcb revision

> 19. No stackup information on Layer 1 or drill leyer

added

> 5. Is there a need to leave those GND vias unmasked on bottom layer?

@gkasprow:
>Only certain vias should be exposed. Vias must be exposed either on both sides or not at all.

mkuklewski, 2019-08-13:
Ok, thank you very much.
During next review I have found following issues:

After placing normal FMC card with shield it is impossible to mount it, please move those components 
[image]
[image]

L2 layer - differential lines are 4 mils apart - which is too close. Can you create a rule check which will check it? 
[image]

Next step will be HyperLynx simulations - power and signal integrity

mkuklewski, 2019-08-16:
Also, after today meeting we have decided to slightly modify grounding scheme.
Please connect "shell" net to mounting hole of SpaceVPX connector - K1 and K2.

tprzywoz, 2019-08-16:
> Ok, thank you very much.
> During next review I have found following issues:
> 
> After placing normal FMC card with shield it is impossible to mount it, please move those components

Done

> 
> L2 layer - differential lines are 4 mils apart - which is too close. Can you create a rule check which will check it?

New rule keeps lines min. 0.3 mm apart (more than 2 x diff gap) - with two exceptions: area under FPGA  and distance to SMD pad (jumpers for optional connection)

> 
> Please connect "shell" net to mounting hole of SpaceVPX connector - K1 and K2.

4 "pads" objects changed to "plated" and connected to "shell"

tprzywoz, 2019-09-17:
Last changes:
added transistor switch to SM STAT (R197 R198 T4)
mechanical changes
added via's and modified polygons for 3P3V, VADJ