elhep/SpaceVPX_FMC_Carrier_3U #18: 1V2 voltage rail

mkuklewski, 2020-01-03:
Hello,
after powering up the device, we found out that R2 resistor is to small - it should be 9k1 which is used as R6 - but why are they both there? 

I think R2 value is the residue to mount IGLOO2 instead of SmartFusion2, because with this voltage is 1V05. Can you create a proper variant?

gkasprow, 2020-01-03:
R2 should be DNP by default. R6 has the correct value. It makes sure that 1V2 core voltage is 1.2V even with a voltage drop on R20.
If the FPGA core voltage drop is too high, the  R62 can be assembled instead of R6 to compensate for all PDN drop.
R2 can be assembled with 9k1 value only when R20 and IC5 are removed during 1.2V rail debugging.
You don't have to start with Hello posting here :)

mkuklewski, 2020-01-07:
Ok, then @tprzywoz please create a proper mounting variant or update schemes with DNP.
If it is possible then make a note on schemes with description when any of them should be used.

tprzywoz, 2020-01-07:
R2 set to 9k1 and not fitted in default variant.