elhep/SpaceVPX_FMC_Carrier_3U #19: Problem with power isolation

mkuklewski, 2020-01-03:
Hi,
during test I have found out that when 3V3_AUX is switched off and 12V is on, LM8610 responsible for 1V2 turns off, because PWR_OVLn line drops below 0,86V.
The issue probably is with MSP430 which drains current from PWR_OVLn line (when it is turned off), but I am not 100% percent sure, as further tests must be done.

gkasprow, 2020-01-03:
Each CMOS circuit contains protection diodes. So if you disable 3V3_AUX, it will pull all pins to GND via an embedded diode. So, this is normal behavior.

mkuklewski, 2020-01-07:
It is a normal behaviour, but it cannot stay like that in our system - a series resistors (or different solutions) are required to allow device operate without 3V3_AUX.
Lines with this issue:
- FMC_P12V0_OVCn
- PG_P1V2
- PG_P1V8
- PG_P2V5
- PG_P3V3
- PWR_OVLn

tprzywoz, 2020-01-07:
My proposition:
[image]

FMC_P12V0_OVCn - is connected only to MSP430 input - 3k6 series resistor.

PG_P1V2 3k6 series resistor. With disable 3V3_AUX, resistor divider gives 1.83 V (we need 1 V for LT8610 enable input)

PG (1.8 V, 2.5 V, 3.3 V) each 39k series resistor. Again divider gives 2.15 V (we need 2.1 V for T3B)

PWR_OVLn: I added MOSFET to allow MSP430 to reset power. PWR_OVL is now active High.

biedam, 2020-01-15:
In the lines PG_P1V8, PG_P2V5 and PG_P3V3 there is also the resistor R107 = 10k which dominates the pull-up. I propose to increase this resistor to 100k.

Also the PWR reset line is unnecessary since the MSP430 can reset the FPGA. Power cycling will be performed by external power supply. This reset line brings additional spigle point failure.

mkuklewski, 2020-01-15:
> In the lines PG_P1V8, PG_P2V5 and PG_P3V3 there is also the resistor R107 = 10k which dominates the pull-up. I propose to increase this resistor to 100k.

The lines PG_P1V8, PG_P2V5 and PG_P3V3 are indeed pulled up by a 10k resistor, but there is a schottky diode between them. Also PG pins have typical pull-down resistance equals to 680 ohms, which should be enough.

A more significant and necessary change are proposed serial resistors. 
I just wonder if MOSFET T6A and T6B and ability to power-cylce by MSP430 are needed.
It seems to me that until we power two cards from one source it doesn't make sense. The 12V latch-up should be detected by the power supply and the rest by using implemented LT8610.

I propose to return to the previous concept and implement a serial resistor, so that MSP430 does not turn off the main power supply (when MSP430 is turned off).

tprzywoz, 2020-01-15:
So I only added resistors 39k serial to each line. Or change them to 68k to keep PG voltage levels higher when MSP430 is turned off?

gkasprow, 2020-01-15:
DOn't use high pullup values. the leakage current may increase with TID.